amirhosseinchoghaei/Passwall #4: ارور دانلود از sourecforge

mhd13671367, 2023-09-27:
فایل های که لینک sourceforge  دارند دانلود نمیشن با همراه اول شاتل و مخابرات امتحان کردم
فقط سه تا لینک sourceforge 
امکان اش نیست از جای دیگه اونا رو دریافت کرد ؟
نسخه اسنبشات

amirhosseinchoghaei, 2023-09-27:
درود ، انقدر opkg update رو بزنید تا بدون ارور اکی بده بعد از اینکه ارور نداد کامند نصب که قرار دادم مخصوص اسنپ شات ران کنید
چاره ای نیست … من با مخابرات هم این مشکل را دارم اما چندبار تلاش کنید اکی میشه

Wasoud, 2023-09-28:
Hi
Here is two way:

1- For this problem can install wireguard package and add new interface by using cloudflare-warp profile. 

OR 

2- simply you can change your DNS from interface > lan > Advanced Settings and use custom DNS 1.1.1.1 Resolve is fix.

mhd13671367, 2023-09-29:
یا dns 1.1.1.1 هم نشد راهنمای تنظیم وایرگارد و cloadflare warp رو می فرمایید؟

amirhosseinchoghaei, 2023-09-29:
عرض کردم جناب دستور opkg update رو انقدر وارد بفرمایید تا بدون ارور تکمیل بشه سپس دستور نصب را وارد کنید ...

closed 2023-10-06 by amirhosseinchoghaei