amirhosseinchoghaei/Passwall #5: Xray

tkmw1985, 2023-10-03:
Hello dear.  Installed successfully but after reboot, the xray core does not start.  Is it downloading again by the script each reboot into the temp folder?

amirhosseinchoghaei, 2023-10-03:
Hi There , What is your Model of your router ?

tkmw1985, 2023-10-03:
mi4a gigabit

amirhosseinchoghaei, 2023-10-03:
IF You install the script Completely , After each reboot Xray-core will be install automatic. just in 3 min after boot ...

tkmw1985, 2023-10-03:
I am trying to understand. How is the script do that downloading it
remotely from another source or locally from router?

tkmw1985, 2023-10-03:
If remotely then there's a problem for our country cuz almost every single
site got blocked.

amirhosseinchoghaei, 2023-10-03:
Its online

tkmw1985, 2023-10-03:
Can we edit script  to download for example from another source which will
be uploaded by myself ? If yes can you point me which script should I edit
and download exactly ipk file to upload it to another source?

amirhosseinchoghaei, 2023-10-04:
Sure ...

https://github.com/amirhosseinchoghaei/Install-Xray-V2ray-On-Passwall-Openwrt/blob/765db44fb56bd917102a436d210ce275522ef7af/owo.sh#L31

* File that you want upload should be : panel.ipk

tkmw1985, 2023-10-04:
Yes, the problem was solved already found two files panel.ipl and zip file
loaded to another source change link in owo script.

closed 2023-10-04 by amirhosseinchoghaei