amirhosseinchoghaei/Passwall #8: نصب روی رزبری

shehnipoor, 2023-10-09:
سلام
من پسوال را روی رزبری نصب کردم

به وای فای که متصل میشم نت از مودم را دریافت و شیر نمیکند

مشکل چیست؟

amirhosseinchoghaei, 2023-10-09:
درود بر شما ریست فکتوری کنید و در اولین گام وای فای دستگاه را کانفیگ و تست بفرمایید ، بعد پکیج های مد نظر رو نصب کنید ...

shehnipoor, 2023-10-09:
البته به وای فاتی رزبری که وصل میشم اینترنت میده
اما ای پی dhcp مودم رو داره میگیره
همین کاری ک گفتید را انجام بدم؟

amirhosseinchoghaei, 2023-10-09:
متوجه نمیشم ... !

الان رزبری پای رو با وای فای به مودم اینترنت وصل میکنید و نتیجه نمیگیرد 
یا اینکه گوشی شما وقتی به رزبری پای وصل میشه هیچ دیتایی ندارید ؟

کدوم حالت ؟

shehnipoor, 2023-10-09:
ببینید
من با گوشی به VPN 2G متصل شدم
رزبری هم با کابل لن به مودم وصله
اینترنت داره
اما داره dhcp رو از مودم میگیره نه از رزبری
ای پی مودم من 192.168.1.2 هستش
وقتی تنظیمات وای فای VPN 2G رو روی گوشی نگاه میکنم میبینم که gatway 192.168.1.2 هستش

در صورتی که ای پی رزبری 192.168.27.1 هستش

shehnipoor, 2023-10-09:
وقتی هم کانکشن گوشی را از dhcp  روی static میزارم و gatway را 192.168.27.1 قرار میدم
اینترنت نمیگیره دیگ

amirhosseinchoghaei, 2023-10-09:
رزبری شما تک پورت لن هستش درسته ؟

shehnipoor, 2023-10-09:
بله

shehnipoor, 2023-10-09:
میخوام از لن نت رو بگیره و از طریق وای فای شیر کنه

amirhosseinchoghaei, 2023-10-09:
مشکل همینجاست تک پورت نمیتونه هم نت بگیره هم دی اپ سی پی بشه مگر بره توی بحث وی لن بندی و اینا که با openwrt من تجربه این کار رو ندارم متاسفانه ....

closed 2023-10-09 by amirhosseinchoghaei

shehnipoor, 2023-10-09:
قرار نیست لن هردو باشه
لن فقط کار دریافت نت رو قراره بکنه و سیستم ها قراره از وای فای dhcp بگیرن

amirhosseinchoghaei, 2023-10-09:
خب DHCP شما غیر فعال هستش و وای فای شما با لن شما ترکیب شده هستش ... باید برید بخش بخش جدا کنید از br-lan خارج کنید وای فای را و یک اینترفیس مستقل تشکیل بدید ... من باز میگم تاحالا این کار رو انجام ندادم اما مشکل شما دقیقا اینجاست که دی اچ سی پی افتاده روی مودم اصلی به خاطر این قضایا

shehnipoor, 2023-10-09:
یک اینترفیس جدید از نوع dhcp روی lan-brig ساختم و مشکل حل شد

ممنونم از شما بابت اسکریپت خوبتون