amirhosseinchoghaei/Passwall #10: Performance issue

hosseinian, 2023-10-17:
Thanks. Everything is working, and installation was straightforward, but I have a performance issue on mi a4 gigabit. The memory seems insufficient to handle the process, and I have randomly crashed and rebooted the router. Do you have any advice to increase the available memory to handle the process?

amirhosseinchoghaei, 2023-10-17:
Hi Mohsen, When You installed The Script ? 
1 month ago ?

hosseinian, 2023-10-17:
I installed it yesterday.

amirhosseinchoghaei, 2023-10-17:
Unfortunately, there is no advice ! 
But in the Latest Update of script , i set the timer to check Passwall is Available or not … if memory kill the passwall , the script Will Reload Passwall Automatically Without Any reboot .
Timer interval : 1 minutes

hosseinian, 2023-10-17:
Yes, by the way, thanks for your useful script.

amirhosseinchoghaei, 2023-10-17:
May i ask you something !?
When Passwall Crashed And you want to reboot , 
Before that ,, remote SSH to openwrt and use this command : 

sh timer.sh

then send me the text results 

thank you 🙏

tkmw1985, 2023-10-21:
May be Zram help you

closed 2023-10-22 by amirhosseinchoghaei