amirhosseinchoghaei/Passwall #12: openwrt luci problem

b3hzad1, 2023-10-31:
سلام دوست عزیز
متاسفانه بعد از اجرای اسکریپت این ارور میاد و دیگه نمیتونم وارد صفحه  openwrt luci بشم. مجبورم روتر رو ریست سخت افزاری کنم. با نسخه های مختلف فرمور هم تست کردم باز همینطوره. ممنون میشم راهنمایی بفرمایید باید چکار کنم.

> /usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.cbi.datatypes' from file '/usr/lib/lua/luci/cbi/datatypes.lua':
> 	/usr/lib/lua/luci/cbi/datatypes.lua:1: unexpected symbol
> stack traceback:
> 	[C]: in function 'loader'
> 	/usr/lib/lua/luci/ccache.lua:70: in function </usr/lib/lua/luci/ccache.lua:61>
> 	[C]: in function 'require'
> 	/usr/lib/lua/luci/passwall/api.lua:8: in main chunk
> 	[C]: in function 'require'
> 	/usr/lib/lua/luci/controller/passwall.lua:5: in main chunk
> 	[C]: in function 'require'
> 	/usr/lib/lua/luci/dispatcher.lua:1136: in function 'createindex'
> 	/usr/lib/lua/luci/dispatcher.lua:1237: in function 'createtree'
> 	/usr/lib/lua/luci/dispatcher.lua:639: in function 'menu_json'
> 	/usr/lib/lua/luci/dispatcher.lua:888: in function 'dispatch'
> 	/usr/lib/lua/luci/dispatcher.lua:483: in function </usr/lib/lua/luci/dispatcher.lua:482>

amirhosseinchoghaei, 2023-10-31:
درود ، مدل روتر چی هست ؟

b3hzad1, 2023-10-31:
xiaomi 4a gigabite

b3hzad1, 2023-10-31:
یکی دوبار شانسی اجرا شد و تونستم بدون مشکل xray رو هم تنظیم کنم که اونم بعد از خاموش روشن کردن روتر دوباره همین ارور اومد

amirhosseinchoghaei, 2023-10-31:
دلیلش اینه که شما از نسخه اصلی openwrt استفاده نمیکنید ! وارد سایت رسمی openwrt بشید و نسخه 22.03.3 یا 22.03.5 را دانلود و نصب نمایید ، هنگام جایگزین کردن فریمور تیک گزینه keep setting را بردارید ...

https://downloads.openwrt.org/releases/22.03.5/targets/ramips/mt7621/openwrt-22.03.5-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin

b3hzad1, 2023-10-31:
دقیقا همین نسخه ها رو از سایت زیر دانلود و نصب کردم و این مشکل رو دارم. اون تیک رو هم برداشتم.

https://firmware-selector.openwrt.org/?version=22.03.4&target=ramips%2Fmt7621&id=xiaomi_mi-router-4a-gigabit

amirhosseinchoghaei, 2023-10-31:
لطفا لینک بالا را که فرستادم دانلود بفرمایید و جایگزین کنید سپس این کامند را اجرا کنید :

amirhosseinchoghaei, 2023-10-31:
rm -f passwallx.sh && wget https://raw.githubusercontent.com/amirhosseinchoghaei/Passwall/main/passwallx.sh && chmod 777 passwallx.sh && sh passwallx.sh

b3hzad1, 2023-10-31:
خیلی ممنون، الان تست میکنم. اطلاع میدم خدمتتان

b3hzad1, 2023-10-31:
متاسفانه باز همون مشکل رو داره. با اون فایل شما و دستوری که فرستادید، امتحان کردم.
فایل گزارش رو هم ضمیمه میکنم شاید به درد بخوره.
[log.txt](https://github.com/amirhosseinchoghaei/Passwall/files/13221553/log.txt)

b3hzad1, 2023-10-31:
روی نسخه 22.03.3 کار کرد و یکم هم استفاده کردم اما بعد از اینکه روتر رو از برق کشیدم و چند دقیقه بعد وصل کردم، دوباره همین ارور پیدا شد. بنظر مشکل از حافظه است حالا یا اینکه بعد از خاموش روشن شدن دستگاه، پاک میشه یا فضای خالی کم داره....

amirhosseinchoghaei, 2023-11-01:
روترتون نسخه اینترنشنال هست یا ورژن چین v2 ؟

b3hzad1, 2023-11-01:
نه این نسخه است، دقیقا شبیه مشخصات اون فیلمی است مه شما گذاشتید. اینه:
Xiaomi Mi Router 4A Gigabit Edition
MediaTek MT7621 ver:1 eco:3

amirhosseinchoghaei, 2023-11-01:
بهم ریموت انی دسک بدید

b3hzad1, 2023-11-01:
بله، الان میفرستم. فقط دوباره مودم رو ریست کردم و الان بصورت دستی passwall رو طبق اون فیلم راهنماتون نصب کردم و فعلا مشکلی نداره ولی هنوز xray رو نصب نکردم. اگر اشکالی نداره خودتون لطفا اونو نصب کنید.

b3hzad1, 2023-11-01:
320405661

amirhosseinchoghaei, 2023-11-01:
فکر کنم ایرانسل هستید لود نمیشه انی دسک

b3hzad1, 2023-11-01:
الان با همراه اول وصل میشم

b3hzad1, 2023-11-01:
وصل شدم

amirhosseinchoghaei, 2023-11-01:
کامل ببندید مجدد باز کنید آی پی تغییر کرده

amirhosseinchoghaei, 2023-11-01:
انی دسک را کامل quit کنید و مجدد باز کنید ...

b3hzad1, 2023-11-01:
چشم

b3hzad1, 2023-11-01:
320405661

amirhosseinchoghaei, 2023-11-01:
روتر را خاموش روشن کنید

b3hzad1, 2023-11-01:
مجدد روشن شد

b3hzad1, 2023-11-01:
خیلی ممنون از وقتی که گذاشتید. بدون مشکل و با نسخه OpenWrt 22.03.5 داره کار میکنه. فکر کنم نکتۀ دیگه این بود که بعد از خاموش،روشن کردن دستگاه باید چند دقیقه منتظر می‌ماندم تا xray مجدداً بارگذاری و ران بشه.

closed 2023-11-01 by b3hzad1