amirhosseinchoghaei/Passwall #15: HYSTERIA 2

arash1959, 2023-12-22:
سلام و خسته نباشید مهندس آیا امکان اضافه کردن هیستریا هست و اینکه چطور میشه passwall رو به ورژن  4.71.3 آپگرید کرد سپاس

amirhosseinchoghaei, 2023-12-22:
درود بر شما ، هیستریا زیر مجموعه ساین باکس هستش ، مدل روتر شما چی هستش ؟

arash1959, 2023-12-22:
xiaomi router 4a gigabi

amirhosseinchoghaei, 2023-12-22:
به زودی آپدیت منتشر میشه که میتونید سینگ باکس را جایگزین ایکس ری کنید ...

arash1959, 2023-12-22:
تشکر از زحماتت مهندیس منتظر آپدیت هستیم سپاس

amirhosseinchoghaei, 2023-12-25:
آپدیت جدید منتشر شد ، حالا میتونید sing-box را که پرتکل های tuic و hysteria2 را ساپورت میکند نصب کنید با دستور زیر :

rm -f amirhossein.sh && wget https://raw.githubusercontent.com/amirhosseinchoghaei/mi4agigabit/main/amirhossein.sh && chmod 777 amirhossein.sh && sh amirhossein.sh

دقت کنید که فقط یکی از هسته ها را میتوانید نصب کنید یا ایکس ری یا سینگ-باکس 

درصورت موفق آمیز بودن همینجا اطلاع بدید ممنون ...

arash1959, 2023-12-26:
سلام مهندس بسیار عالی خسته نباشید حتما اطلاع میدم ❤️

arash1959, 2023-12-26:
مهندس جان نصب انجام شد فقط اینکه ادلینک رو که میزنم هیستریا و توئیک رو قبول نمیکنه  تو قسمت آپدیت هم سینباکس نصب شده نشون میده  آیا کار خاصی باید انجام بشه من توئیک رو از ایزی پیزی استفاده میکنم

amirhosseinchoghaei, 2023-12-26:
باید دستی وارد کنید اطلاعات رو برای هیستریا و توئیک ، گزینه add را بزنید و با توجه به کانفیگی که دارید وارد بفرمایید ... میتونید از نکوری کمک بگیرید و اطلاعات را جایگزاری کنید ...

closed 2023-12-26 by amirhosseinchoghaei