amirhosseinchoghaei/Passwall #18: Passwall Update

freeforlife-star, 2024-01-09:
سلام
من خیلی وقته که نصب کردم روی مودم Xiaomi 4a Gigabit و ازتون تشکر میکنم
فقط اینکه چک کردم و دیدم که آپدیت اومده برای Passwall
چطور میتونم آپدیت کنم و اینکه چ چیزهایی اضافه شده
ممنون میشم راهنمایی کنید

amirhosseinchoghaei, 2024-01-09:
درود بر شما ، آپدیت جدید بیشتر مربوط میشه به رفع باگ های Hysteria2 و نود سابسکرایب ... من آپدیت کردم اما گویا رم بیشتری درگیر میشه ! و این خوب نیست ... 

برای آپدیت هم از این لینک فایل ipk را دانلود کنید و از قسمت software > upload package فرآیند را انجام دهید...
https://github.com/xiaorouji/openwrt-passwall/releases/download/4.72-1/luci-19.07_luci-app-passwall_4.72-1_all.ipk

freeforlife-star, 2024-01-09:
میشه آپدیت و تست کرد و بعد دوباره برگردوند به ورژن قبلی ؟
اگر امکانش هست لینک ورژن قبلی هم بزارید

amirhosseinchoghaei, 2024-01-09:
برای برگردوندن به ورژن قبلی باید 
opkg remove luci-app-passwall

opkg install luci-app-passwall

closed 2024-01-09 by amirhosseinchoghaei