amirhosseinchoghaei/Passwall #19: مشکل نصب

arianaminiss, 2024-01-12:
سلام وقت بخیر روتر شیائومی 4a gigabit مشکل نصب دارم انگار یه پکیج رو نمیتونه دانلود کنه باید چیکار کنم؟
[image]
[image]
[image]

amirhosseinchoghaei, 2024-01-12:
درود ، روی هاست دانلود اختلال هستش برای نصب وی پی ان سیستم را به اشتراک بزارید روی لن تا بتونه دانلود کنه …

arianaminiss, 2024-01-14:
ببخشید یه سوالی داشتم امکانش هست بشه کاری کرد که کلا درخواست های پورت 3074 از تانل رد نشن؟ توی بخش access control دیوایس مورد نظر برای تنظیمات ادد شده این پورته هم مربوطه به یه بازی هست که فقط از همین پورت استفاده میکنه میخوام کلا اگه بشه تنظیم کنم این پورته هیچ ترافیک vpn ای بهش داده نشه امکانش هست؟

amirhosseinchoghaei, 2024-01-14:
سلام
توی قسمت other setting 
سکشن TCP No Redir Ports رو بزنید و قسمت کاستوم پروت مورد نظر را وارد کنید و اینتر رو بزنید و سیو کنید 
در آخر از بخش دی ان اس پسوال clear ipset رو بزنید 

موفق باشید

closed 2024-01-14 by amirhosseinchoghaei