amirhosseinchoghaei/Passwall #20: Kill Switch

Holyerfan, 2024-01-13:
سلام. من نمیخام که ایپی ایرانم لو بره اما بعضی وقتا چک میکنم ایپی ایرانم میفته روی روتر
اپشن کیل سوییچ رو نوشتین که پیشفرض داره ولی برای من اینطور نیست
این اتفاق هم روی لن من میفته
ممنون میشم راهنمایی کنید..
مودمم 4a Gigabyte

Holyerfan, 2024-01-13:
یادم رفت بگم. وقتی چک میکنم روی وایرلس آیپی اوکی هست... ام روی لن ایپی ایران میفته.

amirhosseinchoghaei, 2024-01-13:
درود، دی ان اس استاتیک ست کردید روی شبکه دستگاهی که به لن وصل میکنید … کارت شبکه دستگاهی که به روتر وصل مبکنید دی ان اس و ای پی حالت اتوماتیک باشد

amirhosseinchoghaei, 2024-01-13:
برای رفع هر گونه باگ درصورت نشتی دی ان اس این کار را انجام دهید 

Network > interface > Wan > Advanced

تیک دی ان اس دیفالت را بردارید و این دی ان اس را وارد کنید و گزینه + را بزنید
1.1.1.1

وارد محیط پسوال شده و از قسکت DNS حالت TCP و 8.8.4.4 و remote dns را بزنید سیو کنید و در اخر Clear ipset را بزنید 

موفق باشید …

Holyerfan, 2024-01-13:
وقت بخیر
من ایپی استاتیکی ست نکردم و فقط موم رو به روتر متصل کردم... دقیقا متوجه نشدم که باید چیکار کنم. میتونم از طریق تلگرام باهاتون ر ارتباط باشم!؟

closed 2024-01-14 by amirhosseinchoghaei