amirhosseinchoghaei/Passwall #21: Problem with Adding vless link

aliir74, 2024-01-15:
Hey man,

Thanks for your awesome work!

I installed `openwrt v23.03.3` as you recommended in my `xiaomi 4a gigabyte` router and installed `sing-box` with your script. But I cannot add a valid `vless` link into my node list. It doesn't add the `vless` link and says ` Invalid Share URL Format: vless`.

Do you have any clue what might be the problem?

p.s. I used your script with installing xray-core and it was working fine with the same `vless` link earlier.

amirhosseinchoghaei, 2024-01-15:
درود بر شما ...

هیچ مشکلی در اضافه کردن لینک در سینگ باکس نباید وجود داشته باشد  الان هم تست کردم ! چه ورژنی از پسوال را دارید استفاده میکنید شما ؟

aliir74, 2024-01-16:
`passwall v4.72-2-smartdns-dev`

aliir74, 2024-01-17:
فکر می‌کنم موفق نشده xray-core رو نصب کنه. اسکریپت‌تون رو دیدم ظاهرا این آدرس دیگه وجود نداره:
`https://amir3.space/pass.ipk`
ورژن sing-box رو هم null زده.

amirhosseinchoghaei, 2024-01-17:
عزیز آدرس درست هستش ...

aliir74, 2024-01-17:
[image]

amirhosseinchoghaei, 2024-01-17:
pass.ipk دقیقا از کجا  آمده ؟

closed 2024-01-17 by amirhosseinchoghaei

aliir74, 2024-01-18:
نگاه بالا به پایین‌تون خیلی جالب نیست.
من توی اسکریپت دست نبردم که اگر می‌بردم هم مشکلی نداشت. روی گیت‌هاب کد رو وقتی گذاشتین چنین اجازه‌ای دارین می‌دین.

شما اسکریپت‌تون رو عوض کردین و مقدار اشتباه رو گذاشتین. می‌تونین اینجا رو مشاهده کنید:
https://github.com/amirhosseinchoghaei/Passwall/blob/69c1b6c08d4c1b383628ca77b3085b62df1bde01/passwallx.sh#L172

این هم کامیتی که این تغییر رو دادین:
https://github.com/amirhosseinchoghaei/Passwall/commit/036ba026d719ed51d5325a16cb2177aa7e966b5d

amirhosseinchoghaei, 2024-01-18:
عزیز این قسمت که برای نصب اولیه پسوال هستش و هیچ ربطی به ایکس ری نداره ... اون pass.ipk هم که گذاشته بودم برای پسوال 4.69 بود که منسوخ شده ورژنش و حذف شده از دیتابیس ... شما باید بخش اسکریپت owo.sh توی بخش root را چک بفرمایید و برای مشاهده سورس کد ها به این بخش مراجعه کنید :  (این بخش مخصوص دیوایس هایی هست که فضای دیسک کافی نیست و اسکریپت هنگام نصب تشخیص میده و به صورت خودکار سوییچ میشه روی این اسکریپت و  براتون نصب میکنه ایکس ری و یا سینگ باکس به انتخاب خودتون  روی تمپ اسپیس ...)

https://github.com/amirhosseinchoghaei/Install-Xray-V2ray-On-Passwall-Openwrt

موفق باشید