amirhosseinchoghaei/Passwall #22: قطع شدن اتصال

arianaminiss, 2024-01-19:
سلام وقت بخیر مودم xiaomi 4a gigabit بعد یه مدت اتصال قطع میشه مجدد باید از بخش node برم با یکی از حالت ها از اول کانفیگ رو وصل کنم توی بخش سیستم لاگ لاگ زیر رو بهم میده کسی میدونه باید چیکار کنم؟

> Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.424962] xray invoked oom-killer: gfp_mask=0xcc0(GFP_KERNEL), order=0, oom_score_adj=0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.433557] CPU: 0 PID: 8167 Comm: xray Not tainted 5.10.176 #0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.439496] Stack : 00000000 8072f628 809b0000 807cc414 00000000 00000000 00000000 00000000
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.447904]         00000000 00000000 00000000 00000000 00000000 00000001 82749b18 bf2ae4f6
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.456290]         82749bb0 00000000 00000000 827499c0 00000038 8039e8a4 ffffffea 00000000
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.464675]         827499cc 0000013c 807d4940 ffffffff 82749af8 806fe1e4 80000000 80700000
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.473055]         80856468 00000054 00001000 807cc414 00000018 8040948c 00000000 80990000
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.481436]         ...
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.483893] Call Trace:
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.483994] [<8039e8a4>] 0x8039e8a4
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.489928] [<8040948c>] 0x8040948c
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.493423] [<80007b08>] 0x80007b08
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.496900] [<80007b10>] 0x80007b10
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.500379] [<803839ac>] 0x803839ac
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.503868] [<8007ffe4>] 0x8007ffe4
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.507360] [<80137758>] 0x80137758
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.510866] [<80137fb8>] 0x80137fb8
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.514360] [<801389bc>] 0x801389bc
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.517838] [<801816c8>] 0x801816c8
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.521344] [<804e4cc8>] 0x804e4cc8
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.524821] [<805b6814>] 0x805b6814
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.528298] [<804e4d4c>] 0x804e4d4c
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.531792] [<805b8040>] 0x805b8040
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.535269] [<8005c528>] 0x8005c528
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.538744] [<800bd4f0>] 0x800bd4f0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.542241] [<805b8798>] 0x805b8798
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.545725] [<802047cc>] 0x802047cc
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.549217] [<804df4e8>] 0x804df4e8
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.552725] [<801a8640>] 0x801a8640
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.556227] [<801a8a4c>] 0x801a8a4c
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.559702] [<800b29d8>] 0x800b29d8
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.563181] [<806e1ac0>] 0x806e1ac0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.566658] [<80014578>] 0x80014578
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.570160]
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.571984] Mem-Info:
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.574327] active_anon:7769 inactive_anon:7029 isolated_anon:0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.574327]  active_file:0 inactive_file:89 isolated_file:0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.574327]  unevictable:0 dirty:0 writeback:0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.574327]  slab_reclaimable:586 slab_unreclaimable:4226
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.574327]  mapped:5358 shmem:7793 pagetables:152 bounce:0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.574327]  free:5015 free_pcp:15 free_cma:0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.605491] Node 0 active_anon:31076kB inactive_anon:28116kB active_file:0kB inactive_file:356kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:21432kB dirty:0kB writeback:0kB shmem:31172kB writeback_tmp:0kB kernel_stack:936kB all_unreclaimable? yes
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.628800] Normal free:20060kB min:16384kB low:20480kB high:24576kB reserved_highatomic:0KB active_anon:31076kB inactive_anon:28116kB active_file:140kB inactive_file:400kB unevictable:0kB writepending:0kB present:131072kB managed:120724kB mlocked:0kB pagetables:608kB bounce:0kB free_pcp:60kB local_pcp:0kB free_cma:0kB
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.656973] lowmem_reserve[]: 0 0 0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.660562] Normal: 440*4kB (ME) 159*8kB (UME) 33*16kB (UME) 19*32kB (ME) 13*64kB (ME) 2*128kB (UM) 3*256kB (UME) 1*512kB (E) 1*1024kB (U) 0*2048kB 3*4096kB (U) = 19848kB
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.675963] 7899 total pagecache pages
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.679727] 0 pages in swap cache
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.683107] Swap cache stats: add 0, delete 0, find 0/0
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.688327] Free swap  = 0kB
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.691256] Total swap = 0kB
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.694139] 32768 pages RAM
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.696924] 0 pages HighMem/MovableOnly
Fri Jan 19 13:07:10 2024 kern.warn kernel: [30935.700812] 2587 pages reserved
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.703974] Tasks state (memory values in pages):
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.708678] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.717381] [    644]    81   644      337       25    16384        0             0 ubusd
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.725602] [    645]     0   645      250        8    16384        0             0 askfirst
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.734046] [    680]     0   680      279       14    20480        0             0 urngd
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.742229] [   1150]   514  1150      338       15    16384        0             0 logd
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.750324] [   1204]     0  1204      533       38    20480        0             0 rpcd
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.758391] [   1410]     0  1410      304       13    20480        0             0 dropbear
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.766840] [   1515]     0  1515      666       14    20480        0             0 hostapd
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.775193] [   1516]     0  1516      666       14    20480        0             0 wpa_supplicant
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.784147] [   1536]   101  1536     1111       25    20480        0             0 wpa_supplicant
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.793097] [   1537]   101  1537     1154       38    24576        0             0 hostapd
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.801472] [   1580]     0  1580      455       37    16384        0             0 netifd
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.809711] [   1638]     0  1638      384       25    16384        0             0 odhcpd
Fri Jan 19 13:07:10 2024 kern.info kernel: [30935.817984] [   1771]     0  1771     1039       48    24576        0             0 uhttpd
[... 15 lines cut ...]
Fri Jan 19 13:07:37 2024 user.info : luci: accepted login on / for root from 192.168.30.104
Fri Jan 19 13:07:38 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Fri Jan 19 13:07:44 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Fri Jan 19 13:07:50 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Fri Jan 19 13:08:00 2024 cron.err crond[8312]: USER root pid 23668 cmd sh /root/timer.sh
Fri Jan 19 13:08:00 2024 user.notice check_internet: CORE is OK
Fri Jan 19 13:08:00 2024 user.notice check_internet: PASSWALL IS OK
Fri Jan 19 13:08:04 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
[... cut ...]
Fri Jan 19 13:09:17 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Fri Jan 19 13:09:24 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Fri Jan 19 13:09:36 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Fri Jan 19 13:09:42 2024 daemon.err uhttpd[1771]: netstat: showing only processes with your user ID
Fri Jan 19 13:09:42 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)

amirhosseinchoghaei, 2024-01-19:
درود .

با این کامند اسکریپت را آپدیت کنید : 

rm -f amirhossein.sh && wget https://raw.githubusercontent.com/amirhosseinchoghaei/mi4agigabit/main/amirhossein.sh && chmod 777 amirhossein.sh && sh amirhossein.sh

موفق باشید

closed 2024-01-19 by amirhosseinchoghaei

arianaminiss, 2024-01-19:
انجام شد ولی مشکل هنوز پابرجاست

> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.211440] active_anon:7410 inactive_anon:6736 isolated_anon:0
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.211440]  active_file:0 inactive_file:126 isolated_file:32
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.211440]  unevictable:0 dirty:0 writeback:0
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.211440]  slab_reclaimable:587 slab_unreclaimable:4416
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.211440]  mapped:5259 shmem:7402 pagetables:156 bounce:0
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.211440]  free:3967 free_pcp:21 free_cma:0
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.242822] Node 0 active_anon:29640kB inactive_anon:26944kB active_file:0kB inactive_file:532kB unevictable:0kB isolated(anon):0kB isolated(file):128kB mapped:21036kB dirty:0kB writeback:0kB shmem:29608kB writeback_tmp:0kB kernel_stack:960kB all_unreclaimable? yes
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.266283] Normal free:15868kB min:16384kB low:20480kB high:24576kB reserved_highatomic:0KB active_anon:29640kB inactive_anon:26944kB active_file:396kB inactive_file:0kB unevictable:0kB writepending:0kB present:131072kB managed:120724kB mlocked:0kB pagetables:624kB bounce:0kB free_pcp:88kB local_pcp:72kB free_cma:0kB
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.294371] lowmem_reserve[]: 0 0 0
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.297984] Normal: 501*4kB (UME) 60*8kB (UME) 58*16kB (UME) 54*32kB (UME) 25*64kB (UM) 14*128kB (UME) 9*256kB (UME) 8*512kB (UME) 1*1024kB (U) 0*2048kB 0*4096kB = 15956kB
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.313503] 7590 total pagecache pages
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.317268] 0 pages in swap cache
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.320633] Swap cache stats: add 0, delete 0, find 0/0
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.325856] Free swap  = 0kB
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.328771] Total swap = 0kB
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.331660] 32768 pages RAM
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.334457] 0 pages HighMem/MovableOnly
> Fri Jan 19 19:36:06 2024 kern.warn kernel: [54271.338329] 2587 pages reserved
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.341473] Tasks state (memory values in pages):
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.346172] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.354833] [    644]    81   644      345       29    16384        0             0 ubusd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.363034] [    645]     0   645      250        8    16384        0             0 askfirst
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.371499] [    680]     0   680      279       14    20480        0             0 urngd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.379703] [   1150]   514  1150      338       36    16384        0             0 logd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.387822] [   1204]     0  1204      565       73    20480        0             0 rpcd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.395929] [   1410]     0  1410      304       13    20480        0             0 dropbear
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.404439] [   1515]     0  1515      666       14    20480        0             0 hostapd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.412851] [   1516]     0  1516      666       14    20480        0             0 wpa_supplicant
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.421852] [   1536]   101  1536     1111       28    20480        0             0 wpa_supplicant
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.430836] [   1537]   101  1537     1154       39    24576        0             0 hostapd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.439220] [   1580]     0  1580      455       40    16384        0             0 netifd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.447488] [   1638]     0  1638      384       28    16384        0             0 odhcpd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.455803] [   1771]     0  1771     1050       62    24576        0             0 uhttpd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.464083] [   2287]     0  2287      666       26    16384        0             0 ntpd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.472207] [   2292]   123  2292      328       11    20480        0             0 ntpd
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.480348] [   2596]     0  2596      328        8    16384        0             0 udhcpc
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.488695] [   2600]     0  2600      283       14    16384        0             0 odhcp6c
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.497040] [   6887]     0  6887      309       14    20480        0             0 dropbear
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.505537] [   6888]     0  6888      329       13    20480        0             0 ash
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.513577] [  17971]     0 17971   157758     9122    69632        0             0 xray
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.521701] [  18048]     0 18048   157554     6754    61440        0             0 xray
[... cut ...]
> Fri Jan 19 19:36:06 2024 kern.info kernel: [54271.588978] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=xray,pid=17971,uid=0
> Fri Jan 19 19:36:06 2024 kern.err kernel: [54271.601559] Out of memory: Killed process 17971 (xray) total-vm:631032kB, anon-rss:16016kB, file-rss:0kB, shmem-rss:20472kB, UID:0 pgtables:68kB oom_score_adj:0
> Fri Jan 19 19:36:17 2024 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)

amirhosseinchoghaei, 2024-01-19:
رم کم میارید کانفیگ سنگین زدید روتر نمیکشه ... به نسخه 22.03.3 دانگرید کنید تا رم کم تری درگیر بشه حداقل ...
تیک keep setting را بردارید تا تمیز نصب بشود ...

amirhosseinchoghaei, 2024-01-19:
خیر ، تفاوتی نداره ... از چه کانفیگی دارید استفاده میکنید ؟ TLS داره کانفیگ ؟

arianaminiss, 2024-01-20:
جای ادرس دامین خورده کانفیگ متصل به cdn هست و بدون tls فقط همون تعداد دستگاه ها بالا هست

amirhosseinchoghaei, 2024-01-20:
پینگ دامین رو بگیرید و در قسمت آدرس ،  آی پی به دست آمده را وارد کنید ...