amirhosseinchoghaei/Passwall #24: مشکل سرعت دانلود

omidtb279, 2024-01-29:
سلام من مشکل سرعت دانلود دارم یعنی وقتی از طریق روتر به اینترنت وصل میشم سرعت دانلود کمتر از 4 مگ هست در صورتی که با V2RAYNG و از طریق ویندوز 10 مگ هست به نظر شما علت از چی میتونه باشه ؟

amirhosseinchoghaei, 2024-01-29:
درود ، دی ان اس روی حالت UDP تست کنید ...

MrSaeid007, 2024-01-29:
> درود ، دی ان اس روی حالت UDP تست کنید ...

برای منم همین جوری روی Xiaomi Mi Router 4A Gigabit Edition. فکر کنم پردازنده روتر بیشتر از 30 مگابیت نمی تونه هندل کنه. و بیشتر از این نمیده سرعت بهم. روتر قوی تری نمی شناسید که بشه روی openwrt نصب کرد و پردازنده اش قوی باشه مثلا در حد 5 میلیون.

amirhosseinchoghaei, 2024-01-29:
بهترین روتری که میتونم بهتون پیشنهاد بدم شیائومی مدل AX3000T هستش که سی پی یو و رم عالی داره ... دیجی کالا موجود هستش میتونید تهیه کنید ...

closed 2024-01-30 by amirhosseinchoghaei