amirhosseinchoghaei/Passwall #30: مشکل اتصال

borderlessnet, 2024-02-26:
وقت بخیر؛
من پکیج قدیمیتون رو (فقط نصب xray) روی یک روتر دیگه دارم و خیلی عالی کار میکنه اما در پکیج جدید (هم نصب xray به تنهایی و هم نصب passwall کامل ) بعد از وارد کردن کانفیگ، گوگل با آی پی کانفیگ باز میشه و چک آی پی، درست هست اما متاسفانه سایتهای فیلتر مثل یوتیوب باز نمیشه، اما اپلیکیشن تلگرام دسکتاپ و تلگرام موبایل و همینطور واتساپ باز میشن. همین کانفیگ در روتر قدیمی بدون مشکل کار میکنه
هر دو روتر xiaomi 4a gig و openwrt ورژن 23.03

amirhosseinchoghaei, 2024-02-26:
درود ، دی ان اس هاتون رو چک کنید مشکل از سمت دی ان اس شما هست

closed 2024-02-26 by amirhosseinchoghaei

borderlessnet, 2024-02-27:
با عرض پوزش، این که فرمودید رو هم انجام دادم، از دیروز هم تمام حالتهاش رو با دی ان اس های مختلف و حتی تغییر دی ان اس مودم دارم تست میکنم اما متاسفانه جواب نمیده و به همون وضعیت هست.

amirhosseinchoghaei, 2024-02-27:
لطفا تلگرام پیام دهید