amirhosseinchoghaei/Passwall #31: mi router 4a gigabyte

aalirezamoradii, 2024-02-27:
سلام و وقت بخیر
من با توجه به داکیومنت های که گذاشتید همه ی مراحل رو انجام دادم
حتی vpn هم running میشه ولی همچنان ip ایران رو نشون میده
رم کافی هم وجود داره

کانفیگی که دارم vless سرور شخصی
پشت کلادفلر هست روی پورت ۴۴۳ و tls

amirhosseinchoghaei, 2024-02-27:
درود ، دی ان اس wan را ست کنید روی 1.1.1.1 و پسوال روی tcp remote 8.8.4.4

aalirezamoradii, 2024-02-27:
متاسفانه روتر وقتی داشت فلش می شد کابلش قطع شد و بریک شد
با نرم افزار خود شیاومی تست کردم نشد فریمور رو برگردونم
راه حله دیگه ای داره؟

hesamimam, 2024-03-01:
سلام براس دیبریک ازین لینک کمک بگیرید
https://www.youtube.com/watch?v=SLbkce-M2nE

aalirezamoradii, 2024-03-01:
> سلام برای دیبریک ازین لینک کمک بگیرید https://www.youtube.com/watch?v=SLbkce-M2nE
مرسی بابت اشتراک
ولی این برای زمانیه کیه روتر روش فریمور هست
هرچقد تلاش کردم با فریمور های مختلف نشد
بعد از اینکه فایل رو با tftp یا ابزاری های که معرفی شده به مودم میفرستم led رنگش بنفش میشه
و دیگه هیچ اتفاقی نمیوفته

closed 2024-03-02 by amirhosseinchoghaei