amirhosseinchoghaei/Passwall #32: Can not Download Packages ... Check your internet Connection

rbsdotnet, 2024-03-03:
دردود دوست عزیز
بعد از نصب این خطا ور بهم میده

Configuring kmod-ipt-nat.
passwallx.sh: line 114: kmod-nft-socket: not found
passwallx.sh: line 116: kmod-nft-tproxy: not found
passwallx.sh: line 137: unzip: not found
ls: /etc/init.d/passwall: No such file or directory
 Can not Download Packages ... Check your internet Connection .
در صورتی که اینترنت وصل هست

rbsdotnet, 2024-03-03:
مشکل در عدم وجود opkg install  قبل از خط 114و 116 بود همچنین Package Unzip هم وجود نداشت.
بعد از رفع این مشکل اصلا به V2ray متصل نمیشه و هیچ سایتی رو باز نمیکنه
اما همون کانفیگ در گوشی و کامپیوتر با نرم افزار متصل هست.

amirhosseinchoghaei, 2024-03-03:
سلام ، اینترنتتون محدودیت داره ، با اینترنت دیگری چک کنید

closed 2024-03-03 by amirhosseinchoghaei

rbsdotnet, 2024-03-03:
بعد از ریست مودم هربار در یک قسمتی خطا میده
Configuring kmod-nft-tproxy.
passwallx.sh: line 137: unzip: not found
ls: /etc/init.d/passwall: No such file or directory
 Can not Download Packages ... Check your internet Connection .

amirhosseinchoghaei, 2024-03-03:
عرض کردم مشکلتون چی هست

rbsdotnet, 2024-03-03:
> عرض کردم مشکلتون چی هست

اینترنت فیبرنوری دارم. تمام فایل ها رو دستی دانلود کردم
امکانش هست anydesk به سیستم من بزنید؟ چون هر روشی بود رو تست کردم جواب نداد

rbsdotnet, 2024-03-04:
من نسخه قبل از تغییرات رو تونستم نصب کنم فکر میکنم یه مشکلی توی نسخه جدید هست که باعث چنین خطایی میشه.
بتونم از اولین فرصت بررسی میکنم و مشکل رو گزارش میکنم

amirhosseinchoghaei, 2024-03-04:
جناب مشکلی نیستش اون اروری که شما دریافت میکنید به این خاطر هست اینترنت شما به دلیل محدودیت هایی که تو ساعات مختلف دارد اجازه نمیدهد پکیج ها از دایرکت لینک ها دانلود شود و وقتی که اسکریپت چک میکنه میبینه پسوال نصب نیست به شما پیام میده که اینترنت خودتون رو چک کنید ... مشکلی نیست من کل کد هارو بررسی کردم

xaviersogini, 2024-06-01:
سلام من هم این مشکل رو دارم با نت ها و VPN های مختلف امتحان کردم اما رفع نشد

quitegreensky, 2024-06-03:
من هم اخیرا به این مشکل برخوردم. با سه تا اپراتور تست کردم. اینم بخشی از لاگ

```
Downloading https://downloads.openwrt.org/releases/22.03.4/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.4/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
No packages removed.
Package dnsmasq-full (2.86-16) installed in root is up to date.
Package unzip (6.0-8) installed in root is up to date.
Unknown package 'luci-app-passwall'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency dns2tcp for luci-app-passwall
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-passwall found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-passwall.
Package ipset (7.15-2) installed in root is up to date.
Package ipt2socks (1.1.4-3) installed in root is up to date.
Package iptables-nft (1.8.7-7) installed in root is up to date.
Package iptables-zz-legacy (1.8.7-7) installed in root is up to date.
Package iptables-mod-conntrack-extra (1.8.7-7) installed in root is up to date.
Package iptables-mod-iprange (1.8.7-7) installed in root is up to date.
Package iptables-mod-socket (1.8.7-7) installed in root is up to date.
Package iptables-mod-tproxy (1.8.7-7) installed in root is up to date.
Package kmod-ipt-nat (5.10.176-1) installed in root is up to date.
Package kmod-nft-socket (5.10.176-1) installed in root is up to date.
Package kmod-nft-tproxy (5.10.176-1) installed in root is up to date.
Archive:  iam.zip
  inflating: /lib/upgrade/keep.d/luci-app-passwall
  inflating: /usr/lib/lua/luci/view/passwall/global/status.htm
  inflating: /usr/lib/lua/luci/view/passwall2/global/status.htm
  inflating: /usr/lib64/lua/luci/view/passwall/global/status.htm
  inflating: /usr/lib64/lua/luci/view/passwall2/global/status.htm
ls: /etc/init.d/passwall: No such file or directory
 Can not Download Packages ... Check your internet Connection .
```

amirhosseinchoghaei, 2024-06-03:
سلام به همگی ، باید برخی از پکیج ها را دستی نصب کنید :

https://github.com/xiaorouji/openwrt-passwall/releases/download/4.77-6/passwall_packages_ipk_mipsel_24kc.zip

آنزیپ کرده و پکیج های : 

dns2tcp
dns2socks
tcping
ip2socks
microsocks
simple-obfs
chinadns-ng

دستی آپلود و نصب کنید .

quitegreensky, 2024-06-03:
> سلام به همگی ، باید برخی از پکیج ها را دستی نصب کنید :
> 
> https://github.com/xiaorouji/openwrt-passwall/releases/download/4.77-6/passwall_packages_ipk_mipsel_24kc.zip
> 
> آنزیپ کرده و پکیج های :
> 
> dns2tcp dns2socks tcping ip2socks microsocks simple-obfs chinadns-ng
> 
> دستی آپلود و نصب کنید .

Fixed 👌