amirhosseinchoghaei/Passwall #33: مشکل - تغییر نکردن ای پی

rbsdotnet, 2024-03-04:
درود دوست عزیز
بعد از نصب آی پی تغییر نمیکنه و آی پی ایران رو نمایش میده
اما Youtube و سایر سایت هایی که فیلتر هستن بدون مشکل باز میشن
مشکل از چی هست؟

amirhosseinchoghaei, 2024-03-04:
DNS

closed 2024-03-04 by amirhosseinchoghaei

rbsdotnet, 2024-03-06:
درود
من تمام dns ها رو بررسی کردم جواب نداد
فقط در صورتی جواب میده که tcp default proxy mode در حالت proxy باشه
این مشکل در نسخه جدید وجود داره

[image]

amirhosseinchoghaei, 2024-03-06:
درود ، هم tcp و هم udp باید در حالت proxy باشند ...

rbsdotnet, 2024-03-06:
> درود ، هم tcp و هم udp باید در حالت proxy باشند ...

پیش فرض این بخش فعال نیست ممنون از راهنمایی شما