amirhosseinchoghaei/Passwall #35: After installing x-ray on the tmp folder

AllenfrankX, 2024-03-13:
Good afternoon, I installed everything on tmp folder and after reboot everything disappears, how can I fix this, can you help please?

amirhosseinchoghaei, 2024-03-14:
Hi there , if you install with my script , after reboot the xray will download automatically .

what is your router ?

AllenfrankX, 2024-03-14:
> Hi there , if you install with my script , after reboot the xray will download automatically .
> 
> what is your router ?

Mi4A gigabit edition Sir, I install via tmp folder

amirhosseinchoghaei, 2024-03-14:
rm -f amirhossein.sh && wget https://raw.githubusercontent.com/amirhosseinchoghaei/mi4agigabit/main/amirhossein.sh && chmod 777 amirhossein.sh && sh amirhossein.sh

try this command

AllenfrankX, 2024-03-14:
Also delete x-ray after reboot system

AllenfrankX, 2024-03-14:
Work, thanks

closed 2024-03-14 by AllenfrankX