amirhosseinchoghaei/Passwall #38: problem on ports

3245, 2024-03-21:
hello
i installed and all is fine, just issue i have is, my linksys modem has 4x output lan port, they not have internet, but wifi is fine
should i do any extra setting for one of the ports have internet?

Thanks

3245, 2024-03-21:
if possible please give me your telegram we troubleshoot together

amirhosseinchoghaei, 2024-03-22:
do not use static dns in your lan client ....
use dynamic dns on your pc or any client .

closed 2024-03-22 by amirhosseinchoghaei

3245, 2024-03-23:
منظورتون اینه توی روتر تنطیمی باید انجام بدم؟
چون من داخل کلاینت تنطیمی انجام ندادم و کلن یک تلویزیون رو وصل کردم به روتر از طریق لن که اصلا تنطیمات دی ان اس اینها نداره، اما بازم اینترنت نداره
تنظیماتی باید توی روتر انجام بدم؟
وایفا درست کار میکنه و وی پی ان هم داره، اما هیچکدوم از ۴تا پورتِ روتر اینترنت نداره

تشکر

amirhosseinchoghaei, 2024-03-24:
اگر زمان اتصال به پورت های lan آی پی دریافت میکنید همه چیز درست هستش ، فقط باید دی ان اس رو بزارید روی حالت اتوماتیک
اگر هم که اصلا آی پی دریافت نمیکنید از سمت روتر مشکل چیز دیگری هستش ریست فکتوری کنید

3245, 2024-04-05:
یکسری تصاویر ضمیمه کردم ممکنه چک کنید و بگید آیا درسته این قسمت ها؟ چون واقعا اذیت میکنه🙏❤️
اگر از قسمت دیگه ای نیاز هست بگین عکس بدم
مدل روتر Linksys WRT1900ACS  

[image]
[image]
[image]
[image]
[image]
[image]

amirhosseinchoghaei, 2024-04-05:
سلام همه چیز درست هستش ، موبایل یا لپ تاب از دی ان اس استاتیک استفاده نکنید و روی حالت اتوماتیک باشه