amirhosseinchoghaei/Passwall #46: Fragment

freeforlife-star, 2024-04-29:
سلام
من طبق دستورات Passwall و xray رو با آخرین ورژن روی مودم Xiaomi 4a Gigabit نصب کردم
برای ریالیتی و CDN هایی که هنوز فیلتر نشده اند عالیه و به خوبی کار میکنه

وقتی تیک Fragment رو از قسمت Other Setting میزنم، دامنه های CDN فیلتر با کانفیگ vless/ws/cdn رو URL Test که میگیرم پینگشون رو دارم
ولی رو کامپیوتر اینترنت کلا میره و هیچ جایی دسترسی ندارم، حتی گوگل دیگه باز نمیشه
ولی از تو CMD ، پینگ 8.8.8.8 رو دارم
انگار ی جورایی DNS از کار میوفته

میشه لطفا راهنمایی کنید برای استفاده از Fragment چه گزینه های دیگه ای رو باید تنظیم کنم ؟

amirhosseinchoghaei, 2024-04-29:
سلام ، فرگمن به درستی روی پسوال کار میکنه پارامتر های فرگمنت را با توجه به کانفیگتون رعایت کنید ...

closed 2024-04-29 by amirhosseinchoghaei

freeforlife-star, 2024-04-29:
دقیقا همون تنظیماتی رو میدم که روی گوشیم یا سیستم خونه میدم و اون کانکشن ها کار میکنن
عکس براتون میفرستم
اگر جایی تنظیمی اشتباه هست من درستش کنم
ممنون از شما

[image]

[image]

[image]

[image]

amirhosseinchoghaei, 2024-04-29:
چرا تیک ipv6 رو وشن کردید توی دی ان اس ؟ خاموشش کنید 
tcp 
remote dns

مقادیر فرگمنت هم با سرور باید تطبیق بدید من نمیدونم کانفیگ سرور شما چی هست ، پسوال موردی نداره.