amirhosseinchoghaei/Passwall #48: سلام

omidtb279, 2024-04-29:
سلام من مشکلی که روی روتر 4a نسخه giga  شیائومی دارم این هست که پروکسی هی قطع و وصل میشه راه حلی داری ؟

amirhosseinchoghaei, 2024-04-29:
سلام رم پایینه روتر xiaomi ax 3000t بخرید موفق باشید

closed 2024-04-29 by amirhosseinchoghaei

omidtb279, 2024-04-30:
مدل linksys-ea8300 هم اکی هست بخرم ؟

omidtb279, 2024-05-02:
1- ایا نصب اینکه فروشنده دی جی کالا گفته نسخه اینترنشنال روتر xiaomi ax 3000t دارای  openwrt از پیش نصب شده هست حقیقت داره ؟
2-  آیا نصب openwrt روی روتر xiaomi ax 3000t آسان هست ؟
3- آیا شما آموزشی در مورد نصب openwrt روی این روتر دارین ؟

omidtb279, 2024-05-02:
> پیشنهاد نمیکنم

1- ایا نصب اینکه فروشنده دی جی کالا گفته نسخه اینترنشنال روتر xiaomi ax 3000t دارای openwrt از پیش نصب شده هست حقیقت داره ؟
2- آیا نصب openwrt روی روتر xiaomi ax 3000t آسان هست ؟
3- آیا شما آموزشی در مورد نصب openwrt روی این روتر دارین ؟

amirhosseinchoghaei, 2024-05-02:
تهیه کنید مشکلی نداره ... فقط بعدش ایمیج رسمی را از سایت openert تهیه و جایگزین کنید

omidtb279, 2024-05-09:
> تهیه کنید مشکلی نداره ... فقط بعدش ایمیج رسمی را از سایت openwrt تهیه و جایگزین کنید

ممنون

omidtb279, 2024-05-31:
> تهیه کنید مشکلی نداره ... فقط بعدش ایمیج رسمی را از سایت openwrt تهیه و جایگزین کنید

ممکنه بگین دقیقا کدام نسخه ؟

amirhosseinchoghaei, 2024-05-31:
https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-mediatek-filogic-xiaomi_mi-router-ax3000t-squashfs-sysupgrade.bin