amirhosseinchoghaei/Passwall #52: Rule Management

paytakhtgithub, 2024-05-10:
سلام،
من متوجه شدم علیرغم آپدیت کردن فایل های geosite و geoip رول هایی که بر اساس geosite هستن اعمال نمیشن و به روش دیفالت باهاشون برخورد میشه مثال geosite:playstation
در صورتی که رول هایی که بر اساس دامنه نوشته شدن به خوبی انجام میشن مثال domain:playstation.net
تنظیمات domain strategy هم روی IPifnonMatch هست

amirhosseinchoghaei, 2024-05-11:
درود ، مدل روتر ؟

paytakhtgithub, 2024-05-12:
میکروتیک RB750GR3
Ram: 256MB
CPU: 880MHz

amirhosseinchoghaei, 2024-05-12:
به نظر میاد پکیج های geoip va geosite به دلیل کم بودن حافظه داخلی شما نصب نشده است ، با فلش مموری افزایش حجم انجام دادید ؟

paytakhtgithub, 2024-05-12:
بله با اسکریپت شما و فلش ۸ گیگابایت افزایش دادم
حتی به صورت دستی هم دو تا فایل رو آپدیت کردم و چک کردم با ssh وجود دارن در روتر

amirhosseinchoghaei, 2024-05-12:
opkg update
opkg install v2ray-geoip
opkg install v2ray-geosite

این دستور را اجرا بفرمایید ، در صورتی که این دو پکیج نصب نباشند مشکل شما رفع میشود
اما در صورتی که پکیج ها نصب بوده باشند شما یجای کار رو دارید اشتباه میکنید من از بابت این آپشن مطمن هستم و همین الان در حال حاضر برای من به درستی عمل میکند .

closed 2024-05-12 by amirhosseinchoghaei

paytakhtgithub, 2024-05-12:
هر دو پکیج ها نصب بودن و به روز
اما همچنان سایت های ایرانی باز نمیشن حتی با وجود رول geoip:ir در قسمت ip
[image]

amirhosseinchoghaei, 2024-05-12:
لطفا تلگرام پیام بدید ...