amirhosseinchoghaei/Passwall #53: Error on starting VPN:  iptables+ipset transparent proxy cannot be used on

YounesRahimi, 2024-05-15:
سلام من موقع اجرا کردن VPN  این خطا رو میگیرم:
2024-05-15 21:26:34: The system does not have iptables installed or ipset or Dnsmasq does not have ipset support enabled, so iptables+ipset transparent proxy cannot be used!

لاگ کامل تر و چینی در ادامه:
2024-05-15 21:26:29: The script is already running. Do not run it again and exit.
2024-05-15 21:26:33: Empty and close related programs and cache completed.
2024-05-15 21:26:34: The system does not have iptables installed or ipset or Dnsmasq does not have ipset support enabled, so 
iptables+ipset transparent proxy cannot be used!
2024-05-15 21:26:35: 127.0.0.1#15353 (Direct DNS: 192.168.88.1 Remote DNS: 1.1.1.1)
2024-05-15 21:26:35: - [1] Domain name in node list (vpslist): 192.168.88.1
2024-05-15 21:26:35: - [0]Default: 127.0.0.1#15353
2024-05-15 21:26:43: Restart dnsmasq service
2024-05-15 21:26:44: Configure scheduled tasks: automatic update rules.
2024-05-15 21:26:44: Run completed!

لاگ چینی:
2024-05-15 21:26:29: 脚本已经在运行，不重复运行，退出.
2024-05-15 21:26:33: 清空并关闭相关程序和缓存完成。
2024-05-15 21:26:34: 系统未安装iptables或ipset或Dnsmasq没有开启ipset支持，无法使用iptables+ipset透明代理！
2024-05-15 21:26:35: 127.0.0.1#15353 （直连DNS：192.168.88.1 远程DNS：1.1.1.1）
2024-05-15 21:26:35:   - [1]节点列表中的域名(vpslist)：192.168.88.1
2024-05-15 21:26:35:   - [0]默认：127.0.0.1#15353
2024-05-15 21:26:43: 重启 dnsmasq 服务
2024-05-15 21:26:44: 配置定时任务：自动更新规则。
2024-05-15 21:26:44: 运行完成！

مشخصات سیستم:

Hostname | By-AmirHossein
Model | GL.iNet GL-AR300M (NOR)
Architecture | Qualcomm Atheros QCA9533 ver 2 rev 0
Target Platform | ath79/nand
Firmware Version | OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-22.361.69894-438c598
Kernel Version | 5.10.161

ورژن لایبراری های مربوطه:

dnsmasq-full | 2.86-16 | 176.9 KB | It is intended to provide coupled DNS and DHCP service to a LAN.…

kmod-nf-ipt | 5.10.161-1 | ~19.8 KB | Iptables core

ipset | 7.15-2 | ~2.3 KB | IPset administration utility | Remove…
kmod-ipt-ipset | 5.10.161-1 | ~183.1 KB | IPset netfilter modules | Remove…
libipset13 | 7.15-2 | ~44.6 KB | IPset administration utility

firewall4 | 2022-10-18-7ae5e14b-1 | ~28.8 KB | This package provides an nftables-based implementation of the UCI firewall… 
kmod-nf-flow | 5.10.161-1 | ~16.0 KB | Netfilter flowtable support | Remove…
kmod-nf-ipt | 5.10.161-1 | ~19.8 KB | Iptables core | Remove…
kmod-nft-core | 5.10.161-1 | ~102.3 KB | Kernel module support for nftables | Remove…
kmod-nft-fib | 5.10.161-1 | ~6.3 KB | Netfilter nf_tables fib support | Remove…
kmod-nft-nat | 5.10.161-1 | ~5.2 KB | Netfilter nf_tables NAT support | Remove…
kmod-nft-offload | 5.10.161-1 | ~5.1 KB | Netfilter nf_tables routing/NAT offload support | Remove…
kmod-nft-socket | 5.10.161-1 | ~2.6 KB | Netfilter nf_tables socket support | Remove…
kmod-nft-tproxy | 5.10.161-1 | ~3.3 KB | Netfilter nf_tables tproxy support | Remove…
nftables-json | 1.0.2-2.1 | ~264.4 KB | nftables userspace utility with JSON support | Remove…
parted | 3.4-3 | ~23.6 KB | GNU Parted manipulates partition tables. This is useful for…

[image]

[image]

[image]

با تشکر

amirhosseinchoghaei, 2024-05-15:
سلام خیلی ممنونم که کامل اطلاعات را ارسال میکنید ... لطفا این کد هارو بزنید ببینید نصب هستند یا خیر :

opkg update
opkg install ipset
opkg install iptables
opkg install iptables-legacy
opkg install iptables-mod-conntrack-extra
opkg install ipt2socks
opkg install kmod-ipt-nat
opkg install kmod-nft-socket
opkg install kmod-nft-tproxy
opkg remove dnsmasq
opkg install dnsmasq-full

amirhosseinchoghaei, 2024-05-15:
درصورتی که مشکل رفع نشد ، تلگرام پیام بدید

YounesRahimi, 2024-05-16:
بله مشکل از عدم نصب بعضی از پکیج هایی بود که لیست کردید.
با نصب شون مشکل حل شد.
تشکر

closed 2024-05-16 by YounesRahimi