amirhosseinchoghaei/Passwall #56: سوال در مورد ax3000t

omidtb279, 2024-05-31:
سلام طبق توصیه شما روتر شیائومی ax3000t رو تهیه کردم و کاملا راضی هستم از خریدم و چند تا سوال داشتم 

[image]

1- ایا نسخه opwnwrt که الان نصب هست اخرین نسخه پایداری هست ؟
2- فرق بین Openwrt SNAPSHOT با Openwrt چیست ؟

amirhosseinchoghaei, 2024-05-31:
سلام ، اسنپ شات در بعضی موارد حتی میتونه خیلی بهتر عمل کنه مخصوصا برای این روتر اسنپ شات به خوبی جواب میده .

closed 2024-05-31 by amirhosseinchoghaei

omidtb279, 2024-05-31:
> سلام ، اسنپ شات در بعضی موارد حتی میتونه خیلی بهتر عمل کنه مخصوصا برای این روتر اسنپ شات به خوبی جواب میده .

خیلی ممنون از پاسخگویی شما

[image]
الان روی این نسخه passwall 2 نصب هست و اینکه نسخه اسنپ شات به روز هست ؟ چون داخل سایت دو نسخه هست نمیدونم فرقشون چی هست ؟ 
[image]
](url)

amirhosseinchoghaei, 2024-05-31:
این نسخه ای که برای شما نصبه نسخه کامپایل شده توسط فروشنده هستش و رسمی نیست اما اگر خوب کار میکنه و مشکلی ندارید تغییری ایجاد نکنید به نظر مشکلی نیست .