amirhosseinchoghaei/Passwall #62: طرز وارد کردن کلید ECH

Daariush, 2024-07-17:
سلام
چگونه باید محتویات کلید ECH را برای نود در پسوال وارد کرد؟
مثلا [
            "-----BEGIN ECH CONFIGS-----",
            "AFf+DQBTAAAgACB6W+9vL92rXF3Pjqo39EYyjO+aW5+dBKwV1INoV/1+cwAIAAEA",
            "AQABAAMAIFstLXBxLXNpZ25hdHVyZS1zY2hlbWVzLWVuYWJsZWRdAAA=",
            "-----END ECH CONFIGS-----"
          ]

amirhosseinchoghaei, 2024-07-17:
درود ، همچین امکانی وجود ندارد

closed 2024-07-17 by amirhosseinchoghaei

Daariush, 2024-07-18:
[image]

amirhosseinchoghaei, 2024-07-18:
بله درست میفرمایید در از سینگ باکس هیستریا و توییک این قابلیت وجود داره 

متشکرم

Daariush, 2024-07-19:
خب سوالم این هست که چطوری باید وارد بشه؟ در سوال اصلی کلید را گذاشتم، از کجاش باید گذاشت؟