amirhosseinchoghaei/Passwall #65: عمل نکردن بخش direct list

travelagancy, 2024-08-10:
عرض سلام و احترام خدمت شما دوست گرامی

بنده یه مشکلی دارم، اینکه سایت هایی رو که در بخش direct list قرار میدم تا به صورت مستقیم و با آی پی ایران متصل بشه، گویا کار نمیکنه و تمامی سایت هایی که در بخش direct list هستند، از طریق پروکسی و با آی پی فیلتر شکن متصل میشه.
مودم  Xiaomi Mi 4A Gigabit Edition , و آخرین نسخه passwall  هم نصب کردم.

خیلی ممنون میشم که راهنماییم بفرمایید

amirhosseinchoghaei, 2024-08-10:
ارادت ،
از قسمت basic بخش mode مقادیر را تغییر دهید

travelagancy, 2024-08-10:
تشکر بابت پاسخگویی‌تون🙏🏼🙏🏼

[image]

لطفا میفرمایید که در این بخش mode دقیقا کدوم مقادیر رو باید تغییر بدم تا direct list به درستی عمل کنه؟

amirhosseinchoghaei, 2024-08-10:
اینجا بزارید روی گلوبال پروکسی و سیو کنید همچنین از بخش دایرکت لیست هم عکس بدید چه کانفیگی وارد فرمودین ؟

travelagancy, 2024-08-10:
بخش دایرکت لیست همان سایت هایی هست که به صورت پیشفرض هنگام نصب بود
[image]

به طور مثال از درگاه شاپرک (shaparak.ir) داخل دایرکت لیست قرار داره، اما وقتی از درگاه اینترنتی میخوام خرید کنم،  آی پی رو خارج از ایران تشخیس میده و حتما باید پروکسی رو قطع کنم

amirhosseinchoghaei, 2024-08-10:
https://account.tamin.ir/auth/login

این سایت باز میشه ؟

travelagancy, 2024-08-10:
بله باز میشه👍🏼

amirhosseinchoghaei, 2024-08-10:
پس داره عمل میکنه دامنه شاپرکی که براتون باز نمیشه به صورت دستی اد کنید ، روترتون خیلی ضعیف هستش این روش دایرکت قدیمی میباشد روش بهتری برای دایرکت کردن ایران ترافیک هست که نیازمند رم 256 میباشد ... موفق باشید

closed 2024-08-10 by amirhosseinchoghaei

travelagancy, 2024-08-10:
دست گلتون درد نکنه، یک دنیا سپاس🌸🌸🌸