amirhosseinchoghaei/Passwall #68: دامنه های با آی پی پرایوت

paytakhtgithub, 2024-09-08:
سلام وقت بخیر
علیرغم وجود رول هایی مثل دامنه medu.ir و رنج آی پیِ آی پی های پرایوت سایت medu.ir باز نمیشه و در صورت گرفتن traceroute هم با بد آدرس مواجه میشه
فکر میکنم مربوط به DNS باشه اما راه حلی پیدا نکردم میخواستم نظر شما رو بدونم
متشکرم

amirhosseinchoghaei, 2024-09-08:
ارادت ، دامنه را در قسمت Network , DHCP and DNS 

Domain whitelist

وارد کنید

closed 2024-09-08 by amirhosseinchoghaei

paytakhtgithub, 2024-09-08:
مرسی مهندس جان حل شد
البته rebinding protection رو از dnsmasq کلا خاموش کردم تا دامنه های با آی پی پرایوت ریسالو بشن
پیشنهادتون برای 403 گوگل آنالیتیکس و تگ منیجر چی هست؟
با ست کردن DoH برای مدتی باز میشه اما بعد از گذشت 5 دقیقه دوباره 403 میده

amirhosseinchoghaei, 2024-09-08:
نیازمند آی پی تمیز هستید ،

paytakhtgithub, 2024-09-08:
آی پی اوکیه از وارپ رد میشه، بدون پسوال و کلاینت هایی مثل V2rayN همه چیز اوکیه
سر همون عرض کردم