amirhosseinchoghaei/Passwall #69: Core NOT RUNNING

arash1959, 2024-09-21:
سلام مهندس خسته نباشید امروز x جایگزین اینستا شده در قسمت بیسیک ستینگ ولی مشکل Core NOT RUNNING  پیش امد در حالیکه همه چیز درست کار میکنه و ارتباط وی پی ان برقراره

amirhosseinchoghaei, 2024-09-22:
ارادت ، باگ برطرف شد ممنون از اطلاع شما ، روتر رو یکبار خاموش روشن کنید لطفا ... اگر مشکل پا برجا بود اطلاع بدید .

arash1959, 2024-09-22:
سپاس من هم آپدیت کردم هم روتر رو ریبوت کردم ولی متاسفانه همچنان  مشکل پا برجاست البته مشکلی در کارکرد نیست فقط نات رانینگ نمایش داده میشه

amirhosseinchoghaei, 2024-09-22:
مجدد روتر رو خاموش روشن بفرمایید من تغییرات را اعمال کردم نتیجه را اعلام کنید

arash1959, 2024-09-23:
[image]
[image]

amirhosseinchoghaei, 2024-09-24:
مجدد روتر را خاموش روشن بفرمایید اپدیت جدید اعمال شد

arash1959, 2024-09-24:
[image]
با این آپدیت مشکل حل شد مهندس سپاس از زحمات شما

closed 2024-09-25 by amirhosseinchoghaei