amirhosseinchoghaei/Passwall #70: اضافه کردن هسته مهسا

asghar61, 2024-10-11:
سلام . آقا دمت گرم چه اسکریپت جالبی .میتونی هسته مهسا رو استفاده کنی که از فرگمنت قسمت پکت رو بتونیم tlshello  بزاریم . بازم ممنون بابت کار جالبی که کردی

amirhosseinchoghaei, 2024-10-11:
ارادت ، بله حتما . مدل روترتون را اعلام کنید اینجا

asghar61, 2024-10-11:
دمت گرم . یه ویدئو ازش میزارم کارت خیلی باحال بود .

asghar61, 2024-10-12:
[image]
امیر حسین جان خطا در آپدیت کور مهسا

amirhosseinchoghaei, 2024-10-12:
مدل روتر را اعلام بفرمایید ...

amirhosseinchoghaei, 2024-10-12:
مجدد نصب بفرمایید تغییرات اعمال شد .

rm -f passwallx.sh && wget https://raw.githubusercontent.com/amirhosseinchoghaei/Passwall/main/passwallx.sh && chmod 777 passwallx.sh && sh passwallx.sh

asghar61, 2024-10-12:
مرسی از سرعت عمل . رو پی سی نصبه x86 -64

amirhosseinchoghaei, 2024-10-12:
چک فرمودین ؟ توی قسمت app update ایکس ری رو آپدیت کنید و اگر گفت آپدیته فورس آپدیت را بزنید .

asghar61, 2024-10-12:
باشه - بزار تست کنم

asghar61, 2024-10-12:
[image]
 موقع نصب خطا میده

amirhosseinchoghaei, 2024-10-12:
>/usr/lib/lua/luci/passwall2/com.lua

curl https://raw.githubusercontent.com/amirhosseinchoghaei/Passwall/refs/heads/main/com >>/usr/lib/lua/luci/passwall2/com.lua

این دوتا رو به ترتیب توی ssh بزن

asghar61, 2024-10-12:
قدایی داری  اوکی شد

closed 2024-10-12 by amirhosseinchoghaei