amirhosseinchoghaei/Passwall #72: Failed to download https://master.dl.sourceforge.net/project...

uchar, 2024-11-04:
سلام موقع استفاده از اسکریپت mahsa core + passwall 2 این خطا داده میشه 
``` opkg_download: Failed to download https://master.dl.sourceforge.net/project/openwrt-passwall-build/releases/packages-23.05/aarch64_cortex-a53/passwall_packages/Packages.gz, wget returned 4.```

لاگ کامل

```
- Model : Xiaomi Redmi Router AX6S
 - System Ver : 23.05.3
 - System Arch : aarch64_cortex-a53

 1.  Install Passwall 1
 2.  Install Passwall 2 ( Requires a +256 MB RAM )
 3.  Install Passwall 2 + Mahsa Core
 9.  CloudFlare IP Scanner
 6.  EXIT

 -Select Passwall Option : 3
Installing Passwall 2 With Mahsa Core ...
Downloading 'https://raw.githubusercontent.com/amirhosseinchoghaei/Passwall/refs/heads/main/mahsa.sh'
Connecting to 2606:50c0:8003::154:443
Writing to 'mahsa.sh'
mahsa.sh             100% |*******************************|  5232   0:00:00 ETA
Download completed (5232 bytes)
Running as root...
 Updating Packages ...
Downloading https://master.dl.sourceforge.net/project/openwrt-passwall-build/releases/packages-23.05/aarch64_cortex-a53/passwall_luci/Packages.gz
Updated list of available packages in /var/opkg-lists/passwall_luci
Downloading https://master.dl.sourceforge.net/project/openwrt-passwall-build/releases/packages-23.05/aarch64_cortex-a53/passwall_luci/Packages.sig
Signature check passed.
Downloading https://master.dl.sourceforge.net/project/openwrt-passwall-build/releases/packages-23.05/aarch64_cortex-a53/passwall_packages/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from https://master.dl.sourceforge.net/project/openwrt-passwall-build/releases/packages-23.05/aarch64_cortex-a53/passwall_packages/Packages.gz

Downloading https://master.dl.sourceforge.net/project/openwrt-passwall-build/releases/packages-23.05/aarch64_cortex-a53/passwall2/Packages.gz
Updated list of available packages in /var/opkg-lists/passwall2
Downloading https://master.dl.sourceforge.net/project/openwrt-passwall-build/releases/packages-23.05/aarch64_cortex-a53/passwall2/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download https://master.dl.sourceforge.net/project/openwrt-passwall-build/releases/packages-23.05/aarch64_cortex-a53/passwall_packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

Downloading 'https://master.dl.sourceforge.net/project/openwrt-passwall-build/passwall.pub'
Connecting to 216.105.38.12:443
Redirected to /project/openwrt-passwall-build/passwall.pub on downloads.sourceforge.net
Redirected to /project/openwrt-passwall-build/passwall.pub?viasf=1 on yer.dl.sourceforge.net
Writing to 'passwall.pub'
[... 82 lines cut ...]
Installing kmod-nf-socket (5.15.150-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/kmod-nf-socket_5.15.150-1_aarch64_cortex-a53.ipk
Configuring kmod-nf-socket.
Configuring kmod-nft-socket.
Installing kmod-nft-tproxy (5.15.150-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/kmod-nft-tproxy_5.15.150-1_aarch64_cortex-a53.ipk
Installing kmod-nf-tproxy (5.15.150-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/kmod-nf-tproxy_5.15.150-1_aarch64_cortex-a53.ipk
Configuring kmod-nf-tproxy.
Configuring kmod-nft-tproxy.
Package ca-bundle (20230311-1) installed in root is up to date.
Installing kmod-inet-diag (5.15.150-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/kmod-inet-diag_5.15.150-1_aarch64_cortex-a53.ipk
Configuring kmod-inet-diag.
Package kernel (5.15.150-1-03c9d191cd600eae3e66845031612f0a) installed in root is up to date.
Installing kmod-netlink-diag (5.15.150-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/kmod-netlink-diag_5.15.150-1_aarch64_cortex-a53.ipk
Configuring kmod-netlink-diag.
Installing kmod-tun (5.15.150-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/kmod-tun_5.15.150-1_aarch64_cortex-a53.ipk
Configuring kmod-tun.
ls: /etc/init.d/passwall2: No such file or directory
 Can not Download Packages ... Check your internet Connection .

```

amirhosseinchoghaei, 2024-11-04:
سلام اینترنت رو بزارید روی همراه اول و از دی ان اس 1.1.1.1 استفاده کنید

closed 2024-11-04 by amirhosseinchoghaei

uchar, 2024-11-20:
> doesn't works

سلام برای من همراه اول کار کرد نیازی به تغییری دی ان اس هم نبود

mika2463, 2024-11-22:
not working also