amirhosseinchoghaei/Passwall #78: facebook

amiras1981, 2024-12-20:
باسلام از تقربا یک هفته پیش فیسبوک و ایکس با روتر که passwall روش نصب هست باز نمیشه اما بصورت مستقیم با اتصال به سرور xray باز میشه و مشکلی نداره آیا باید تنظیمات خاصی در روتر انجام بدیم

mika2463, 2025-01-03:
من هم همین مشکل رو دارم روتر شیاومی mi4a gigabyte داشتم و الان یک ماه میشه پسوالش از کار افتاده پسوال هم نصب کردم اما باز هم وی پی ان استابلش نمیشه مجبور شدم یه 3000t بگیرم ببینیم اگه روی اون کار کنه یا نه کلا اگه میدونستم روی mi4a از کار میفته هزینه نمیکردم مستقیم همین 3000t میگرفتم توی یه issue دیگه توضیح دادم چجوری میشه دوباره پسوال نصب کرد افا امیرحسین هم خودش که روش داده اونم جواب میده اما پسوال اینیبل نمیشه.

pmrk70, 2025-01-04:
سلام بنده مشکلی مشابه دارم تلگرام مثل هلو باز میشه ولی یوتیوب نه جالبه لوکیشن میگیرم آیپی سرور خارج رو نشون میده ولی در عمل یوتیوب باز نمیشه فیسبوک باز نمیشه هم لوکیشن آلمان تست کردم هم انگلیس لوکیشن عوض میشه،چیزی که من قبلا هم به آقا امیر حسین گفته بودم که Xray رو آپدیت میکنم بعد از ریبوت میره روی نسخه قبل دفعه پیش خواستم بروز کنم اجازه نداد ولی وقتی ca-bundle رو آپدیت کردم xray هم بروز شد ولی مجدد بعد از ریبوت میپره الآن من نسخه paswall 25.1.1 هست ولی Xray 1.8.7 که بصورت دستی به 25.1.1 بروزش میکنم مشکلی هم نداشتم چند ماهه همین روال رو انجام میدم اوکیه احساس میکنم باگ passwall باشه اگه آقا امیرحسین کاری بتونه بکنه ممنون میشم جوابمون رو بده

amirhosseinchoghaei, 2025-01-04:
سلام وقت بخیر در پسوال گزینه Mode را بزنید سپس بر روی گزیه Global Proxy کلیک کرده و سیو کنید ...

amirhosseinchoghaei, 2025-01-04:
سپس یکبار مودم را خاموش روشن کنید .

pmrk70, 2025-01-04:
انجام دادم ولی تغییری نکرد یه چیز دیگه که هست تست پینگ میگیرم از ابرآروان و... همه رو problem detected میزنه ولی وقتی از https://whatismyipaddress.com/ تست میگیرم لوکیشن تغییر کرده ولی پینگ نمیده،تلگرام رو باز میکنه ولی یوتیوب رو نه فیسبوک رو نه اینستا رو هم باز میکنه اصلا خیلی عجیب غریبه

amirhosseinchoghaei, 2025-01-04:
مدل های مختلف DNS را چک کنید توی بخش dns پسوال

pmrk70, 2025-01-05:
فایده نداره من حتی چک کردم از بخش software همه رو آپدیت کردم تمام dns هارو هم تست کردم فایده نداره،مسئله ای که هست اگر مشکل از دستگاه باشه کلا نباید کار کنه نه اینکه فقط چندتا سایت مشکل داشته باشه

amirhosseinchoghaei, 2025-01-05:
تلگرام پیام بدید

pmrk70, 2025-01-05:
پیام دادم به اسم DigiAi

amiras1981, 2025-01-05:
با سلام به دوستان من یه ماری کردم و مشکل حل شد نوع کانکشن xray که وصل نمیشد به بعضی از سایتها vmes یا vles بود چند نوع دیگه رو امتحان کردم نهایتا روی shadowsocks که تست کردم مشکل حل شد و فیسبوک و توییتر و سایتهایی که باز نمیشد باز شد با سرعت بالا

amirhosseinchoghaei, 2025-01-05:
مشکل با نصب کردن هسته SingBox برطرف شد ،

closed 2025-01-05 by amirhosseinchoghaei

pmrk70, 2025-01-05:
دوستان از کجا میتونم یه کانفیگ با کیفیت بگیرم؟