amirhosseinchoghaei/Passwall #81: فعال/غیر فعال نشدن سرویس paswall از طریق luci web

travelagancy, 2025-02-13:
سلام و احترام خدمت شما دوست گرامی

من روتر گوگل AC-1304 رو خریداری کردم،  از طریق luci web و چک باکس main switch نمیتونم سرویس passwall  رو فعال یا غیر فعال کنم، و حتما باید از طریق ترمینال و ssh به صورت دستی دستور `service passwall start` یا `service passwall stop` رو بزنم تافعال و غیر فعال بشه.

[image]

هم passwall رو نصب کردم این مشکل وجود داشت و هم passwall 2.

تمامی ورژن های openwrt از 23.05.0 تا 24.10.0 رو نصب کردم و متاسفانه روی تمای ورژن ها اینمشکل وجود داشت.

خیلی ممنون میشم راهنماییم بفرمایید🙏🏼🙏🏼

amirhosseinchoghaei, 2025-02-13:
سلام وقتی تیک مین سوییچ را برمیدارید یا میزارید باید از قسمت دی ان اس پسوال دکمه کلیر آی پی را بزنید ، پیشنهاد میکنم از پسوال 2 استفاده کنید تا از این باگ ها در امان باشید.

closed 2025-02-13 by amirhosseinchoghaei

travelagancy, 2025-02-13:
تشکر بابت پاسخگوییتون💐💐

از قسمت DNS هم دکمه clear رو زدم اما باز هم متاسفانه از طریق luci web و گزینه main swtich نمیشد پسوال رو فعال و غیر فعال کرد

[image]

بعدش طبق فرمایش‌تون پسوال 2 رو نصب کردم و همچنان این مشکل که از طریق وب نمیشه فعال و غیر فعال کرد، روی پسوال 2 هم پا برجا بود.

[image]

amirhosseinchoghaei, 2025-02-13:
مدل روتر و همچنین ورژِن سیستم عامل را ارسال بفرمایید

travelagancy, 2025-02-13:
router: google wifi AC-1304 (Gale)

openwrt 23.05.5

mostafaafrouzi, 2025-02-15:
من هم دقیقا روی همین روتر و همین ورژن OpenWRT این مشکل دارم.
حتما باید در ترمینال /etc/init.d/passwall2 restart
بزنم تا تغییرات اعمال شود

hosseinc, 2025-03-11:
Same here. With google ac 1304